USCRPL/mbed-cmake #2: CMake cannot find `arm-none-eabi-g++`

ProExpertProg, 2020-06-21:
On a linux system (Ubuntu 18.04 WSL).

I installed all the dependencies, and am performing step 7 of [this guide](https://github.com/USCRPL/mbed-cmake/wiki/Project-Configuration).

This is the error I'm getting.

```
CMake Error at mbed-cmake/CMakeLists.txt:35 (project):
  The CMAKE_C_COMPILER:

    arm-none-eabi-gcc

  is not a full path and was not found in the PATH.

  Tell CMake where to find the compiler by setting either the environment
  variable "CC" or the CMake cache entry CMAKE_C_COMPILER to the full path to
  the compiler, or to the compiler name if it is in the PATH.

CMake Error at mbed-cmake/CMakeLists.txt:35 (project):
  The CMAKE_CXX_COMPILER:

    arm-none-eabi-g++

  is not a full path and was not found in the PATH.

  Tell CMake where to find the compiler by setting either the environment
  variable "CXX" or the CMake cache entry CMAKE_CXX_COMPILER to the full path
  to the compiler, or to the compiler name if it is in the PATH.
```

I've added the `bin` folder of the Arm toolchain to the path in `.bashrc`, `.bash_profile`, and `.profile`.

I've added the following lines in cmake, just above `include(mbed_gcc_arm_toolchain)`:
```
...
message(STATUS "PATH:")
message(STATUS $ENV{PATH})

include(mbed_gcc_arm_toolchain)

project(mbed-cmake LANGUAGES C CXX ASM)
...
```

Which print
```
-- PATH:
-- /usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games
```

This makes me think `cmake` only has access to `.profile`, which is reasonable to me. Still, the PTH clearly contains the path to the Arm toolchain; if I perform

```
$ ls /usr/share/gcc-arm/bin
```
I get:
```
arm-none-eabi-addr2line  arm-none-eabi-elfedit    arm-none-eabi-gcc-ranlib     arm-none-eabi-gdb-add-index-py  arm-none-eabi-objcopy  arm-none-eabi-strip
arm-none-eabi-ar         arm-none-eabi-g++        arm-none-eabi-gcov           arm-none-eabi-gdb-py            arm-none-eabi-objdump
arm-none-eabi-as         arm-none-eabi-gcc        arm-none-eabi-gcov-dump      arm-none-eabi-gprof             arm-none-eabi-ranlib
arm-none-eabi-c++        arm-none-eabi-gcc-9.3.1  arm-none-eabi-gcov-tool      arm-none-eabi-ld                arm-none-eabi-readelf
arm-none-eabi-c++filt    arm-none-eabi-gcc-ar     arm-none-eabi-gdb            arm-none-eabi-ld.bfd            arm-none-eabi-size
arm-none-eabi-cpp        arm-none-eabi-gcc-nm     arm-none-eabi-gdb-add-index  arm-none-eabi-nm                arm-none-eabi-strings
```

Googling the error itself did not give me any useful feedback. I'd appreciate any pointers!

jayasurya-sridharan, 2020-06-21:
What's the result when you run ` which arm-none-eabi-g++`? Just to make sure that the compiler is actually in your path.

multiplemonomials, 2020-06-21:
Also is this from CLion, or from the command line?

ProExpertProg, 2020-07-12:
Both command line and CLion; they had different PATHs but they both contained `/usr/share/bin`.

When I run `which arm-none-eabi-g++`, I get `/usr/local/bin/arm-none-eabi-gcc`.

I resolved this issue by setting the compilers explicitly using `-DCMAKE_C_COMPILER=/usr/local/bin/arm-none-eabi-gcc -DCMAKE_CXX_COMPILER=/usr/local/bin/arm-none-eabi-g++`. In CLion, that can be achieved by setting the toolchain directly.

When you set the toolchain directly, make sure that `include(mbed_gcc_arm_toolchain)` happens before any `project()` calls; otherwise, you might get an error in terms of 
```
The C compiler
    "/usr/local/bin/arm-none-eabi-gcc"
  is not able to compile a simple test program.
```
which happens because CMake is testing the compiler before you set the required options.

closed 2020-07-12 by ProExpertProg

ProExpertProg, 2020-07-12:
While the solution above works, it fails later in the build process because it fails to find other `arm-none-eabi-*` files.

A better solution is making sure that the correct path gets added to `PATH`. This issue might be unique to CLion on Windows using WSL:

When CLion runs CMake, it does not read `.bashrc`. Instead, it only gets its `PATH` from `/etc/environment`. Hence, if you add the `bin` directory of `arm-none-eabi` to the definition there, CMake will be able to access all of the toolchain executables.

Be careful: `/etc/environment` contains important path definitions (to things like `ls`,`cp`,`mount`), so if you mess it up, you won't be able to access them.

Thank you @multiplemonomials and @jayasurya-sridharan!

Narzog, 2021-08-19:
I have tried all the solutions above and have the same errors and it still wouldn't recognize the compiler.