USCRPL/mbed-cmake #6: Always build .bin & .hex

ladislas, 2020-07-24:
With https://github.com/ARMmbed/mbed-os/pull/13011, mbed-os tools now build both `.bin` and `.hex`.

I've added the following to `MBedExecutable.cmake`:

```cmake
  # generate .hex firmware file
	add_custom_command(
		TARGET ${EXECUTABLE} POST_BUILD
		COMMAND arm-none-eabi-objcopy -O ihex $<TARGET_FILE:${EXECUTABLE}> ${HEX_FILE}
		WORKING_DIRECTORY ${CMAKE_CURRENT_BINARY_DIR}
		COMMENT "Objcopying ${EXECUTABLE} to mbed firmware ${HEX_FILE}")
```

It is the best way to do it? If so, I can provided a PR. :)

multiplemonomials, 2020-07-25:
Looks good to me!  Hang on until after I merge (or give up on) the Arm Compiler support branch though.  
Just curious, what exactly is the hex file useful for?

multiplemonomials, 2020-07-26:
OK feel free to submit a pr against master.

ladislas, 2020-07-27:
While working on the PR, I was wondering if it would be easier to arm-gcc and armc logic in separate directories/files.

There are already a lot of if/else and if we want to support other compilers in the future, it's only going to be more and more complicated.

The check for `MBED_TOOLCHAIN_NAME` should only happen once at the beginning and then everything else should be relative to that.

In the case of the PR #10, I've also made the changes to generate an .hex with armc but I don't know if it's useful or usable.

I'm also happy to work on separating the different compilers.

> Just curious, what exactly is the hex file useful for?

- .bin files are required for updating the device
- .hex files are very handy for post build (external to build system) custom
scripts (e.g. image signing)

I use a .hex with openocd to flash an external QSPI flash memory.

multiplemonomials, 2020-07-28:
> While working on the PR, I was wondering if it would be easier to arm-gcc and armc logic in separate directories/files.

Yeah I am keeping an eye on this, I already did do that with one file (the mbed-src CMakeLists.txt).  If it gets much worse I'll try to start splitting it out a bit more.

commit a1709980 (2020-09-15): Merge pull request #10 from ladislas/generate-hex-file

Generate .hex file from .elf - closes #6
commit dc9a7808 (2020-09-15): Generate .hex file from .elf - closes #6
closed 2020-09-15 by multiplemonomials