USCRPL/mbed-cmake #15: mbed-cmake or mbed with cmake?

ladislas, 2020-09-16:
Hi guys! 

I'm currently working on implementing our build and CI system.

Following PR https://github.com/ARMmbed/mbed-os/pull/13566 it seems the mbed team is seriously working on the topic even though the number of targets supported is very limited, they are lacking a lot of common options from mbed-cli.

I've been playing with mbed-cmake and I really like it. I haven't run into any issues at the moment, and I'm working on project template with mbed-cmake integrated.

I guess the question is:

- should we keep using mbed-cli for the moment?
- should we use mbed-cmake?
- should we wait for mbed to finish their cmake implementation? (but that can take some serious time)

I don't mind the solution, I just don't want to have to change our build and CI system in a near future because one of those is not supported anymore.

Thanks for you help!

Best,  
-- Ladislas

multiplemonomials, 2020-09-16:
I actually just posted a thing about this to the mbed-cmake forum thread!

ladislas, 2020-09-21:
For people asking the same question, here's @multiplemonomials's detailed answer on ARM Mbed Forums

> https://forums.mbed.com/t/mbed-cli-alternative-mbed-cmake/9357/20