USCRPL/mbed-cmake #16: Pass custom profile.json

ladislas, 2020-09-18:
When using mbed-cli, I pass a custom profile with additional options like this:

```
mbed compile -m DISCO_ORIGINAL --profile=release --profile=./profiles/leka.json
```

How can I do the same with mbed-cmake? (it must be super simple but I can seem to find it...)

multiplemonomials, 2020-09-21:
Currently this doesn't exist sadly.  It would require some modifications to configure_for_target.py.  I am wondering, is there anything that you can do with a profile.json that you can't also do by adding flags in CMake?

ladislas, 2020-09-21:
> I am wondering, is there anything that you can do with a profile.json that you can't also do by adding flags in CMake?

No, that's what I figured and what I'll do! Thanks for the answer :)

closed 2020-09-21 by ladislas