USCRPL/mbed-cmake #18: Add the ability to not provide an upload method

ladislas, 2020-09-24:
Let say I just want to build my code and not use an upload method, it would be nice to have the ability to leave the method to `none` and still compile.

Or is it needed somewhere else?

jasperswallen, 2020-09-24:
I'm able to build bin files without setting an upload method, just make sure you use `make target` and not `make flash-target`.

ladislas, 2020-09-25:
@jasperswallen thanks for the input!

I'm actually able to build without uploading (I'm using Ninja).

My issue is when I'm first running the configuration (it was not clear in my first post), I must be missing something...

ladislas, 2020-09-25:
I found a fix by adding

```cmake
set(UPLOAD_METHOD_DEFAULT NONE)
```

At the beginning of https://github.com/USCRPL/mbed-cmake/blob/master/cmake/UploadMethods.cmake#L5

multiplemonomials, 2020-09-28:
(note that this should actually go in your [UploadMethodConfig.cmake](https://github.com/USCRPL/mbed-cmake-example-project/blob/master/cmake/UploadMethodConfig.cmake), at least if you use the example project)

closed 2020-09-28 by multiplemonomials