USCRPL/mbed-cmake #20: Simpler toolchain setup for Linux

ProExpertProg, 2020-10-12:
As explained in #2, the toolchain instructions for Linux are incomplete, as they don't correctly add everything to your path. 

It is much easier to simply install them with `apt`:

```sh
$ sudo apt update -y
$ sudo apt-get install -y gcc-arm-none-eabi
```

I think the docs should list this as the default installation method and provide the current one as backup. I've tested this on Ubuntu 20.04 using WSL and CLion and it works.

ProExpertProg, 2021-02-06:
Also found a better way to install on Mac: https://github.com/ARMmbed/homebrew-formulae

```sh
brew tap ArmMbed/homebrew-formulae
brew install arm-none-eabi-gcc
```

multiplemonomials, 2021-02-06:
Added to the wiki page!  https://github.com/USCRPL/mbed-cmake/wiki/Toolchain-Setup

Thanks for the tips

closed 2021-02-06 by multiplemonomials

jasperswallen, 2021-02-07:
Also, using choco with Windows makes this much easier as well: `choco install gcc-arm-embedded`