USCRPL/mbed-cmake #22: Usage with custom targets - custom_target.json not recognized

mjbogusz, 2021-02-08:
Hi,
first, thanks for the great project - not having to hack away on the mbed-generated cmake files is a blessing!

However, one thing that I didn't manage to get working properly is custom targets - the wiki only mentions this:
> If you are using an MBed-enabled processor on your own board, you will usually want to use the name of the dev board for that processor, though sometimes the processor name itself will work.

I've tried declaring the target by creating the `custom_target.json` file and the `TARGET_MY_CUSTOM_TARGET` directory as I did for the mbed-cli, but it wasn't picked up regardless of the placement of these files.

The only way I was able to get my custom target to get recognized by the `configure_for_target.py` script was to place the target's directory in the `mbed-src/targets` tree and add it in the main `targets.json` file.
This is not ideal as I can't simply add these changes to a VCS.
My workaround was using `git diff --patch` and `git apply` wrapped in a script and keeping the resulting patch in my VCS, but this has the downside of having to run the scripts on EVERY new environment (as it includes the `PinNames.c`, `PeripheralNames.h` and so on), not only for reconfiguration.

Is there something I've missed? A specific combination of `custom_target.json` and `TARGET_MY_CUSTOM_TARGET` placement? An argument to the `configure_for_target.py` script?

multiplemonomials, 2021-03-01:
Can you try out the `mbed-pathless` branch?  I integrated a change which adds a `-x` option to configure_for_target.py.  You should be able to pass the directory containing custom_targets.json to that option and it should load.

multiplemonomials, 2021-03-26:
This has now been merged into master.

closed 2021-03-26 by multiplemonomials

mjbogusz, 2021-04-12:
Finally I've got the time to test this out and while the `configure_for_target.py` script recognizes the `custom_targets.json` file now, it completely ignores my `TARGET_MY_CUSTOM_TARGET` directory whatsoever.

This means that my target's source files don't get added to the `MbedOSConfig-TARGET.cmake` file. The only workaround I've managed to make work is quite a roundabout one:
* Manually add a cmake target (`INTERFACE` library) for each board
* Add a `target_include_directories` for the `cmake-os-static` target (this will get complicated quickly if targeting multiple boards)
* Link my executable target against my custom board target
* Add any files my custom board overwrites (e.g. `system_clock.c`) to `.mbedignore` (And if one custom target overwrites a file, all others have to do it too)

If there is a better way to do it, please let me know. In the meantime, I'm trying to figure out how to modify the `configure_for_target` script to utilize the custom target's directory.

multiplemonomials, 2021-04-13:
Oh you know, I just ran into the same thing a few weeks ago, and ended up with a hacky CMake fix similar to what you did.  Your PR seems like a way better solution.