USCRPL/mbed-cmake #23: debugging with openocd

mczerski, 2021-03-01:
How I should debug with openocd ?
Runing make debug-<target> fails with connection timeout to gdb server. If I run openocd manually and them make debug-<target> it works as expected.
Shouldn't the make debug-<target> run also openocd in the background ?

multiplemonomials, 2021-03-01:
From the wiki:
> To debug, first run `make gdbserver` in one terminal to start up the debug probe server. Then, you can connect to the GDB server in your debug tool of choice, or run `make debug-xxx` in another terminal to start command-line GDB on the given program.

It would be kinda cool if it could run gdbserver automatically, but I don't think cmake can start 2 processes at the same time in a single target?

mczerski, 2021-03-01:
ok, thanks for clarification.
As for running two processes, You can always wrap multiple commands into some script.

multiplemonomials, 2021-03-01:
ok glad things are working!

closed 2021-03-01 by multiplemonomials