USCRPL/mbed-cmake #24: Including modules?

sinfloodmusic, 2021-03-05:
Hi,

I'm running a simple HelloWorld and trying to include "EthernetInterface.h" but it can't be found.

fatal error: EthernetInterface.h: No such file or directory
    3 | #include "EthernetInterface.h"

I see references to mbed-netsocket in the project, I think that's what I need, but I'm not sure the best way to add both the sources and the includes and make sure everything links together nicely.

Do I just add add them to the "add_mbed_executable" function in mbed-cmake/cmake/MbedExecutable.cmake?

target_link_libraries(${EXECUTABLE} mbed-os mbed-netsocket)

Thank you and great work on this project!

multiplemonomials, 2021-03-06:
Ah I think you might be getting confused between the new Mbed official build system and mbed-cmake.  mbed-cmake is a monolithic build so all enabled Mbed OS modules are under the mbed-os target.  Only the official cmake build system does the legwork to split Mbed out into modules.

Probably the reason why you can't include it is the default mbedignore file.  Try deleting mbed-cmake/mbed-src/.mbedignore, then rerun configure_for_target.py.  That should get things running!

sinfloodmusic, 2021-03-07:
Ahh yes, I was getting confused there.  That worked for me in a fresh directory, but not in my existing project.  

FYI I found that clearing my top level build directory and blowing out the .mbedignore was enough to help the build find ```EthernetInterface.h``` but ultimately it ran into the following linking error:

```
gcc-arm-none-eabi-9-2020-q2-update/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/bin/ld: cannot find -lmbed-netsocket
```

This was just trying to to build the example UDP code here https://os.mbed.com/docs/mbed-os/v6.5/apis/udpsocket.html

However, a completely fresh clone of mbed-make and re-configuring got me up and running.

Thank you for your help!

multiplemonomials, 2021-03-07:
ok glad it's working!

closed 2021-03-07 by multiplemonomials