USCRPL/mbed-cmake #25: support for black magic probe upload method

mczerski, 2021-03-16:
Hi,
It would be great if BMP were supported. Currently I'm using BMP with OpenOCD upload method but I need to modify mbed-cmake.gdbinit generated file to be able to debug.

multiplemonomials, 2021-03-16:
Hmm, I can look into this.  Does this probe have its own uploader and/or gdb server?  What are the modifications you need to make?

mczerski, 2021-03-17:
BMP runs its own gdbserver (on the probe itself) and comunicates over usb serial. So to use it I just start gdb and then connect to it with:
```
target extended-remote <path to serial device like ttyACM0>
```
and then to connect to the mcu I do:
```
monitor swdp (scan for connected devices)
attach <n> (connect to n'th device)
```
After that I can load/debug/or whatever. So basically my .gdbinit file has those three lines.
So to use debug-<target_name> target I need to change mbed-cmake.gdbinit file to have those three lines in it.
The flash-<target_name> is not usable with BMP now since it uses openocd. To support flashing with BMP the flash target would  need to use gdb (connect/load/exit).

multiplemonomials, 2021-03-17:
Hmm, that sounds like it could be tough because every build target would need its own gdbinit.  Hmm maybe there's a way to pass commands to gdb on the command line instead.

mczerski, 2021-03-17:
maybe --eval-command and --init-eval-command options may help with this ?