USCRPL/mbed-cmake #32: Automatic upload not working on WSL

ProExpertProg, 2021-05-05:
Running this through CLion 2021.1, CMake 3.16, latest mbed-cmake (1.6.2). Build and manual upload works successfully, but the flash-target gives the following error:

```
[1620181867.73][mbedls.platform_database]Error loading database /home/luka/.local/share/mbedls/platforms.json: Platform Database is out of date; Recreating
[1620181867.73][mbedls.lstools_linux]Could not get disk devices by id. This could be because your Linux distribution does not use udev, or does not create /dev/disk/by-id symlinks. Please submit an issue to github.com/armmbed/mbed-ls.
[1620181867.73][mbedls.lstools_linux]Could not get serial devices by id. This could be because your Linux distribution does not use udev, or does not create /dev/serial/by-id symlinks. Please submit an issue to github.com/armmbed/mbed-ls.
Traceback (most recent call last):
  File "/mnt/c/Users/Luka Govedic/CLionProjects/dennis-firmware/hardware/lib/mbed-cmake/cmake/upload_methods/install_bin_file.py", line 60, in <module>
    targets = get_detected_targets()
  File "/mnt/c/Users/Luka Govedic/CLionProjects/dennis-firmware/hardware/lib/mbed-cmake/cmake/upload_methods/install_bin_file.py", line 22, in get_detected_targets
    detect_muts_list = mbeds.list_mbeds()
  File "/home/luka/.local/lib/python3.8/site-packages/mbed_os_tools/detect/lstools_base.py", line 135, in list_mbeds
    candidates = list(self.find_candidates())
  File "/home/luka/.local/lib/python3.8/site-packages/mbed_os_tools/detect/linux.py", line 54, in find_candidates
    usb_info = self._sysfs_block_devices(disk_ids.values())
  File "/home/luka/.local/lib/python3.8/site-packages/mbed_os_tools/detect/linux.py", line 121, in _sysfs_block_devices
    sysfs_block_devices = set(os.listdir(SYSFS_BLOCK_DEVICE_PATH))
FileNotFoundError: [Errno 2] No such file or directory: '/sys/class/block'
make[3]: *** [hardware/hardware-test/CMakeFiles/flash-stats.dir/build.make:57: hardware/hardware-test/CMakeFiles/flash-stats] Error 1
make[2]: *** [CMakeFiles/Makefile2:246: hardware/hardware-test/CMakeFiles/flash-stats.dir/all] Error 2
make[1]: *** [CMakeFiles/Makefile2:253: hardware/hardware-test/CMakeFiles/flash-stats.dir/rule] Error 2
make: *** [Makefile:170: flash-stats] Error 2
```

What could be wrong?

multiplemonomials, 2021-05-05:
I think this is because WSL doesn't implement the full USB device API.  WSL's access to hardware is pretty limited in general...

ProExpertProg, 2021-05-06:
In the old version of mbed-cmake, I was actually able to mount the drive and set up automatic uploading.

How exactly does it mbed-cmake scan for devices?

multiplemonomials, 2021-05-06:
Oh right, this is because I switched to using the Mbed Tools python script to do the uploading.  It's nice because it scans the list of USB devices and identifies Mbed boards automatically.  I guess it doesn't work with WSL though.

If you really have to work under WSL (native Windows development has always worked fine for me), your best bet would be to copy the [old version of the MBED upload script](https://github.com/USCRPL/mbed-cmake/blob/a974f8903244fdd38e95d14d51639a56de1273f4/cmake/upload_methods/UploadMethodMBED.cmake) into your top-level project.  Then rename MBED to something else (`HARDCODED_MBED`?) in the file and its filename, and finally enable it in your upload method config file (`set(HARDCODED_MBED_UPLOAD_ENABLED)`) and on the command line (`-DUPLOAD_METHOD=HARDCODED_MBED`).

ProExpertProg, 2021-05-07:
OK. I mean it's just a copy operation, we can just call it manual_copy, right?

ProExpertProg, 2021-05-12:
So I implemented this back but it's currently not working. Have you ever had any problems with the `file` command in CMake? If I copy over the file manually in the terminal, it works

ProExpertProg, 2021-05-12:
So after some more debugging, I think it might be an issue with CMake. If I run the following in the terminal:

```bash
# doesn't work
cmake -E copy "/mnt/c/Users/Luka Govedic/CLionProjects/dennis-firmware/cmake-build-debug-hardware/hardware/hardware-test/clock.bin" "/mnt/d"`

# works
cp "/mnt/c/Users/Luka Govedic/CLionProjects/dennis-firmware/cmake-build-debug-hardware/hardware/hardware-test/clock.bin" "/mnt/d"`
```
The CMake `copy` command works for non-mounted paths. However, it stops the current executable so something is happening (which is even weirder).

I'm inclined to just use the native `cp`; that's because this upload only really applies to WSL.

```cmake
# file(COPY "${BIN_FILE}" DESTINATION "${MBED_PATH}") # doesn't work
execute_process(COMMAND cp "${BIN_FILE}" "${MBED_PATH}") # works
```

multiplemonomials, 2021-05-12:
Could you make it so that on Linux and Mac it will use cp, and on Windows it will use cmake -E copy?

ProExpertProg, 2021-05-12:
Sure, if I knew how to detect WINDOWS haha

multiplemonomials, 2021-05-12:
Oh that's easy: https://gitlab.kitware.com/cmake/community/-/wikis/doc/tutorials/How-To-Write-Platform-Checks
See the section I wrote there, "Platform Checking"

multiplemonomials, 2021-11-07:
Interesting, I'm actually seeing the same behavior on Windows 10 with a NUCLEO_F429ZI board.  Calling cp from Git Bash works (as well as dragging the file in from Explorer), but using file(COPY) seemingly causes the board to hang.  Looking back, I know I've used the `file(COPY)` method with Nucleo boards on Linux and with LPC1768 boards on Windows and Linux, but I'm not sure I'd ever tried using `file(COPY)` on Windows with an ST board, so maybe that's an issue.

multiplemonomials, 2021-11-07:
Resolved (for the most part) in #33.

closed 2021-11-07 by multiplemonomials