USCRPL/mbed-cmake #35: error while running linker script

Michael-Liao, 2021-07-26:
I have configured the example project as stated in the tutorial. My target is `NUCLEO_G031K8`. The cmake generation is fine with no errors if I give `-D UPLOAD_METHOD=STM32CUBE`. However, it gives a warning about not finding `intelhex` and `prettytable` but I am sure I have them installed. In fact, they came with the mbed-cli package.

When I run `make hello_world`, it gives this warning.
```
28:2: error: #error "MBED_ROM_SIZE not defined"
   28 | #error "MBED_ROM_SIZE not defined"
      |  ^~~~~
```

But I did check the outputs of cmake, MBED_ROM_SIZE was set in the CXX compiler flags.

I also suspected it's because I installed my python packages in a virtual environment (`pipenv` to be exact). But then I installed globally and still it doesn't find the dependencies.

multiplemonomials, 2021-08-02:
Are you using Arm Compiler or GNU?  Also can you post a pastebin of the full build output?

multiplemonomials, 2021-11-07:
Ah, I was able to reproduce this.  Turns out there was an issue where some of the arguments to the linker script preprocessor were getting quoted when they shouldn't've been, causing them to be ignored (!).  I guess NUCLEO_G031K8 is the first target to be used that actually uses those defines.

multiplemonomials, 2021-11-07:
Also note that I needed
```
"requires": ["bare-metal"]
```
in my mbed_app.json to link any programs for this target, as the full mbed OS seems to use too much RAM and makes the linker script error out.

commit 36f4e9a8 (2021-11-07): Fix command for preprocessing linker script to pass args correctly (fixes #35)
closed 2021-11-07 by multiplemonomials