USCRPL/mbed-cmake #37: Error in check for working C compiler - "Can't compile a simple test program" (Apple Silicon M1)

aaroexxt, 2021-09-27:
Hi,

I'm trying to use the mbed-cmake repo to compile some code for a student team at my college.
However, when I try to compile it, I get some weird compiler issues - I seem to have no issues with the path, but some of the flags passed to the ARM compiler by mbed-cmake appear to be invalid based on the debug message below.
My current understanding of gcc is that -arch is an OSX-specific flag that isn't actually supported by gcc itself - could this be part of the issue?

I'm using CLion 2021.2.2, CMake 3.21.3, mbed-cmake version 1.4.1, arm-none-eabi-gcc version 10.3.1 20210621 (release) and python 3.9.6 to run the scripts, as well as the following compiler flags in CLion:
```
-DHARDWARE_TARGET=True
-DUPLOAD_METHOD=MBED
-DMBED_PATH=/Volumes/NOD_H743ZI2
```
(DMBED_PATH points to the local STM32 Nucleo board I'm uploading to)
I'm also running all of this on my Macbook Air, which uses a M1 ARM chip - perhaps that's causing some issues.

Let me know if there's anything to try to resolve this issue - I'd really appreciate it!

The following is a full dump from a single attempt to build from CLion:

```
-- Building firmware
--  _       _    ____      ______   ___                      ____    _       _         _        _      _   ______
-- | \     / |  /  _  \   /  ____) /  _ \_                 / ____)  | \     / |       / \      / \    / | /  ____)
-- |  \   /  |  | |_|  |  | |____  | | \_ \    ______    /  /       |  \   /  |      / _ \     | |   / /  | |____
-- |   \ /   |  |     /   |  ____) | |   | |  (______)  |  |        |   \ /   |     / /_\ \    |  _ / /   |  ____)
-- | |\ - /| |  |  _   \  | |      | |  _| |             \  \       | |\ - /| |    / _____ \   |  _  \    | |
-- | | \_/ | |  | |_|   | | |____  | |_/ _/                \ \___   | | \_/ | |   / /     \ \  | |  \ \   | |____
-- \_/     \_/  \_____ /  \______) \___ /                   \____)  \_/     \_/  |_/       \_| \_/   \_/  \______)
--
-- mbed-cmake version 1.4.1, running on CMake 3.21.3
-- Checking for Python package pyocd -- not found
-- Could NOT find STLINKTools (missing: STM32CubeProg_PATH)
-- Board upload method set to MBED
-- The C compiler identification is GNU 10.3.1
-- The CXX compiler identification is GNU 10.3.1
-- Checking whether C compiler has -isysroot
-- Checking whether C compiler has -isysroot - yes
-- Checking whether C compiler supports OSX deployment target flag
-- Checking whether C compiler supports OSX deployment target flag - no
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - failed
-- Check for working C compiler: /usr/local/bin/arm-none-eabi-gcc
-- Check for working C compiler: /usr/local/bin/arm-none-eabi-gcc - broken
CMake Error at /usr/local/Cellar/cmake/3.21.3/share/cmake/Modules/CMakeTestCCompiler.cmake:69 (message):
  The C compiler

    "/usr/local/bin/arm-none-eabi-gcc"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: ~My Code Directory Here (redacted)~/cmake-build/CMakeFiles/CMakeTmp

    Run Build Command(s):/usr/bin/make -f Makefile cmTC_706ff/fast && /Library/Developer/CommandLineTools/usr/bin/make  -f CMakeFiles/cmTC_706ff.dir/build.make CMakeFiles/cmTC_706ff.dir/build
    Building C object CMakeFiles/cmTC_706ff.dir/testCCompiler.c.o
    /usr/local/bin/arm-none-eabi-gcc   -Wall -Wextra -Wno-unused-parameter -Wno-missing-field-initializers -fmessage-length=0 -fno-exceptions -ffunction-sections -fdata-sections -funsigned-char -fomit-frame-pointer -Os -g -DMBED_TRAP_ERRORS_ENABLED=1 -mcpu=cortex-m7 -mthumb -mfpu=fpv5-d16 -mfloat-abi=softfp -DAPPLICATION_ADDR=0x8000000 -DAPPLICATION_SIZE=0x200000 -DAPPLICATION_RAM_ADDR=0x24000000 -DAPPLICATION_RAM_SIZE=0x80000 -DMBED_ROM_START=0x8000000 -DMBED_ROM_SIZE=0x200000 -DMBED_ROM1_START=0x8100000 -DMBED_ROM1_SIZE=0x100000 -DMBED_RAM_START=0x24000000 -DMBED_RAM_SIZE=0x80000 -DMBED_RAM1_START=0x24000000 -DMBED_RAM1_SIZE=0x80000  -arch x86_64 -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX11.3.sdk -std=gnu11 -o CMakeFiles/cmTC_706ff.dir/testCCompiler.c.o -c ~My Code Directory Here (redacted)~/cmake-build/CMakeFiles/CMakeTmp/testCCompiler.c
    arm-none-eabi-gcc: error: x86_64: No such file or directory
    arm-none-eabi-gcc: error: unrecognized command-line option '-arch'; did you mean '-march='?
    make[1]: *** [CMakeFiles/cmTC_706ff.dir/testCCompiler.c.o] Error 1
    make: *** [cmTC_706ff/fast] Error 2

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  firmware/lib/mbed-cmake/cmake/MbedOS-GCCArm.cmake:3 (project)
  firmware/lib/mbed-cmake/mbed-src/CMakeLists.txt:8 (include)
```

multiplemonomials, 2021-09-27:
Does it work if you run cmake from the command line?  It kinda seems like CLion might be adding some extra flags.

multiplemonomials, 2021-09-27:
Also btw I have no access to a Mac so I have a pretty limited ability to test this.

ladislas, 2021-09-27:
Can you share your project? I'll try it from macOS.

aaroexxt, 2021-09-27:
Thank you both for your fast response!

@ladislas Here is a link to the full project directory I'm using - note: quite a large directory at 700MB
[Google Drive link](https://drive.google.com/drive/folders/15Vuk2bEKDSFpGrPzxoDB2md5KiTrmAzt?usp=sharing)
Note: 'cmake-build-debug' and 'cmake-build-debug-hardware' are from CLion, cmake-build is from me trying the command line version as @multiplemonomials mentioned.

@multiplemonomials Also tried running cmake from the command line - output below
Note: CMakeOutput.log and CMakeError.log are attached in the Google Drive link [found here](https://drive.google.com/drive/folders/108f7cRtVZ00vvgpXY_pnIHXOHF4a165w?usp=sharing)

```
cd ~/Desktop/Code/RocketTeam/avionics-onboarding/cmake-build && cmake -S ..

-- Building firmware
--  _       _    ____      ______   ___                      ____    _       _         _        _      _   ______
-- | \     / |  /  _  \   /  ____) /  _ \_                 / ____)  | \     / |       / \      / \    / | /  ____)
-- |  \   /  |  | |_|  |  | |____  | | \_ \    ______    /  /       |  \   /  |      / _ \     | |   / /  | |____
-- |   \ /   |  |     /   |  ____) | |   | |  (______)  |  |        |   \ /   |     / /_\ \    |  _ / /   |  ____)
-- | |\ - /| |  |  _   \  | |      | |  _| |             \  \       | |\ - /| |    / _____ \   |  _  \    | |
-- | | \_/ | |  | |_|   | | |____  | |_/ _/                \ \___   | | \_/ | |   / /     \ \  | |  \ \   | |____
-- \_/     \_/  \_____ /  \______) \___ /                   \____)  \_/     \_/  |_/       \_| \_/   \_/  \______)
--
-- mbed-cmake version 1.4.1, running on CMake 3.21.3
-- Checking for Python package pyocd -- not found
-- Could NOT find STLINKTools (missing: STM32CubeProg_PATH)
-- Board upload method set to MBED
-- The C compiler identification is GNU 10.3.1
-- The CXX compiler identification is GNU 10.3.1
-- Checking whether C compiler has -isysroot
-- Checking whether C compiler has -isysroot - yes
-- Checking whether C compiler supports OSX deployment target flag
-- Checking whether C compiler supports OSX deployment target flag - no
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - failed
-- Check for working C compiler: /usr/local/bin/arm-none-eabi-gcc
-- Check for working C compiler: /usr/local/bin/arm-none-eabi-gcc - broken
CMake Error at /usr/local/Cellar/cmake/3.21.3/share/cmake/Modules/CMakeTestCCompiler.cmake:69 (message):
  The C compiler

    "/usr/local/bin/arm-none-eabi-gcc"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: /Users/Aaron/Desktop/Code/RocketTeam/avionics-onboarding/cmake-build/CMakeFiles/CMakeTmp

    Run Build Command(s):/usr/bin/make -f Makefile cmTC_9566c/fast && /Library/Developer/CommandLineTools/usr/bin/make  -f CMakeFiles/cmTC_9566c.dir/build.make CMakeFiles/cmTC_9566c.dir/build
    Building C object CMakeFiles/cmTC_9566c.dir/testCCompiler.c.o
    /usr/local/bin/arm-none-eabi-gcc   -Wall -Wextra -Wno-unused-parameter -Wno-missing-field-initializers -fmessage-length=0 -fno-exceptions -ffunction-sections -fdata-sections -funsigned-char -fomit-frame-pointer -Os -g -DMBED_TRAP_ERRORS_ENABLED=1 -mcpu=cortex-m7 -mthumb -mfpu=fpv5-d16 -mfloat-abi=softfp -DAPPLICATION_ADDR=0x8000000 -DAPPLICATION_SIZE=0x200000 -DAPPLICATION_RAM_ADDR=0x24000000 -DAPPLICATION_RAM_SIZE=0x80000 -DMBED_ROM_START=0x8000000 -DMBED_ROM_SIZE=0x200000 -DMBED_ROM1_START=0x8100000 -DMBED_ROM1_SIZE=0x100000 -DMBED_RAM_START=0x24000000 -DMBED_RAM_SIZE=0x80000 -DMBED_RAM1_START=0x24000000 -DMBED_RAM1_SIZE=0x80000  -arch x86_64 -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX11.3.sdk -std=gnu11 -o CMakeFiles/cmTC_9566c.dir/testCCompiler.c.o -c /Users/Aaron/Desktop/Code/RocketTeam/avionics-onboarding/cmake-build/CMakeFiles/CMakeTmp/testCCompiler.c
    arm-none-eabi-gcc: error: x86_64: No such file or directory
    arm-none-eabi-gcc: error: unrecognized command-line option '-arch'; did you mean '-march='?
    make[1]: *** [CMakeFiles/cmTC_9566c.dir/testCCompiler.c.o] Error 1
    make: *** [cmTC_9566c/fast] Error 2

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  firmware/lib/mbed-cmake/cmake/MbedOS-GCCArm.cmake:3 (project)
  firmware/lib/mbed-cmake/mbed-src/CMakeLists.txt:8 (include)

-- Configuring incomplete, errors occurred!
See also "/Users/Aaron/Desktop/Code/RocketTeam/avionics-onboarding/cmake-build/CMakeFiles/CMakeOutput.log".
See also "/Users/Aaron/Desktop/Code/RocketTeam/avionics-onboarding/cmake-build/CMakeFiles/CMakeError.log".
```

multiplemonomials, 2021-09-27:
Experiment: What if you try passing `-DCMAKE_SYSTEM_NAME=Generic` to cmake?

multiplemonomials, 2021-09-27:
(you might have to start with a clean build dir)

aaroexxt, 2021-09-27:
Slightly different error, appears to be missing a system library:

```
CMake Error at /usr/local/Cellar/cmake/3.21.3/share/cmake/Modules/CMakeTestCXXCompiler.cmake:62 (message):
  The C++ compiler

    "/Library/Developer/CommandLineTools/usr/bin/CC"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: /Users/Aaron/Desktop/Code/RocketTeam/avionics-onboarding/cmake-build/CMakeFiles/CMakeTmp

    Run Build Command(s):/usr/bin/make -f Makefile cmTC_e2f36/fast && /Library/Developer/CommandLineTools/usr/bin/make  -f CMakeFiles/cmTC_e2f36.dir/build.make CMakeFiles/cmTC_e2f36.dir/build
    Building CXX object CMakeFiles/cmTC_e2f36.dir/testCXXCompiler.cxx.obj
    /Library/Developer/CommandLineTools/usr/bin/CC    -MD -MT CMakeFiles/cmTC_e2f36.dir/testCXXCompiler.cxx.obj -MF CMakeFiles/cmTC_e2f36.dir/testCXXCompiler.cxx.obj.d -o CMakeFiles/cmTC_e2f36.dir/testCXXCompiler.cxx.obj -c /Users/Aaron/Desktop/Code/RocketTeam/avionics-onboarding/cmake-build/CMakeFiles/CMakeTmp/testCXXCompiler.cxx
    Linking CXX executable cmTC_e2f36
    /usr/local/Cellar/cmake/3.21.3/bin/cmake -E cmake_link_script CMakeFiles/cmTC_e2f36.dir/link.txt --verbose=1
    /Library/Developer/CommandLineTools/usr/bin/CC CMakeFiles/cmTC_e2f36.dir/testCXXCompiler.cxx.obj -o cmTC_e2f36
    ld: library not found for -lSystem
    clang: error: linker command failed with exit code 1 (use -v to see invocation)
    make[1]: *** [cmTC_e2f36] Error 1
    make: *** [cmTC_e2f36/fast] Error 2

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  unit-test/CMakeLists.txt:1 (project)
```

aaroexxt, 2021-09-27:
Update: I was able to build my code successfully by adding the following commands to my CMakeLists.txt file:

```
set(CMAKE_SYSTEM_NAME Generic)
set(CMAKE_SYSTEM_PROCESSOR arm)

set(CMAKE_CXX_COMPILER "arm-none-eabi-gcc")
set(CMAKE_CXX_COMPILER "arm-none-eabi-g++")

if(NOT CMAKE_CXX_COMPILE_OBJECT)
    SET(CMAKE_CXX_COMPILE_OBJECT "<CMAKE_CXX_COMPILER> <DEFINES> <FLAGS> -o <OBJECT> -c <SOURCE>")
endif(NOT CMAKE_CXX_COMPILE_OBJECT)

if(NOT CMAKE_CXX_LINK_EXECUTABLE)
       set(CMAKE_CXX_LINK_EXECUTABLE "<CMAKE_CXX_COMPILER>  <FLAGS> <CMAKE_CXX_LINK_FLAGS> <LINK_FLAGS> <OBJECTS>  -o <TARGET> <LINK_LIBRARIES>")
endif()

set(CMAKE_FIND_ROOT_PATH_MODE_PROGRAM NEVER)
set(CMAKE_FIND_ROOT_PATH_MODE_LIBRARY ONLY)
set(CMAKE_FIND_ROOT_PATH_MODE_INCLUDE ONLY)
set(CMAKE_FIND_ROOT_PATH_MODE_PACKAGE ONLY)

set (CMAKE_LINKER_FLAGS "--specs=nosys.specs" CACHE INTERNAL "")
```

However there's a new issue, with the following output after running make:
```
In.cpp:18:10: fatal error: drivers/AnalogIn.h: No such file or directory
   18 | #include "drivers/AnalogIn.h"
```

aaroexxt, 2021-09-28:
Update:
Resolved all mbed-cmake issues by adding `set(CMAKE_SYSTEM_NAME Generic)` to the CMakeLists.txt file as @multiplemonomials suggested - my bad as my configuration had some other issues.
It appears to not break compatibility with Intel-based Macs as well so maybe this is something worth adding into CMakeLists.txt permanently?

closed 2021-09-28 by aaroexxt