USCRPL/mbed-cmake #38: Problem when updating to version 1.8.0

ProExpertProg, 2021-11-16:
Got the following message from CMake when updating to the 1.8.0:

```
-- mbed-cmake version 1.8.0, running on CMake 3.16.3
-- Detected CLion IDE, will generate CLion debug configurations
-- Board upload method set to MBED
CMake Error at hardware/lib/mbed-cmake/cmake/MbedOS-GCCArm.cmake:13 (separate_arguments):
  separate_arguments given unknown argument -Wl,--gc-sections -Wl,-n
  -mcpu=cortex-m7 -mthumb -mfpu=fpv5-d16 -mfloat-abi=softfp
  -DMBED_APP_START=0x8000000 -DMBED_APP_SIZE=0x200000
  -DMBED_ROM1_START=0x8100000 -DMBED_ROM1_SIZE=0x100000
  -DMBED_ROM_START=0x8000000 -DMBED_ROM_SIZE=0x200000
  -DMBED_RAM1_START=0x24000000 -DMBED_RAM1_SIZE=0x80000
  -DMBED_RAM_START=0x24000000 -DMBED_RAM_SIZE=0x80000
  -DMBED_BOOT_STACK_SIZE=1024 -DXIP_ENABLE=0 --specs=nosys.specs
  -Wl,--start-group -lstdc++ -lsupc++ -lm -lc -lgcc -lnosys -Wl,--end-group
Call Stack (most recent call first):
  hardware/lib/mbed-cmake/mbed-src/CMakeLists.txt:8 (include)
```

Running on WSL, `gcc-arm` is newest version (`gcc version 9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599] (15:9-2019-q4-0ubuntu1)
`

Any ideas?

multiplemonomials, 2021-11-16:
Oops looks like I made a typo, but somehow it worked when I was testing on Windows.  Should be fixed now.

closed 2021-11-16 by multiplemonomials

ProExpertProg, 2021-11-16:
Thanks for the super-quick response! Glad it wasn't just me making a stupid mistake 😅

commit c99739eb (2021-11-16): Fix separate_argument() error (which only shows up on Linux for some reason?).  Closes #38