USCRPL/mbed-cmake #40: mingw32-make got Error1 && Error2

nasihs, 2021-11-17:
I followed the steps in Getting Started.
Everything worked smoothly until I tried **mingw32-make -j2 hello_world**.
The outputs are as follows
It would be very nice of you if you could help.
[image]
[image]

multiplemonomials, 2021-11-17:
Hmm, this might be due to path length limits on Windows, I've seen issues with that before.  Try installing [ninja-build](https://ninja-build.org/) (just drop the exe somewhere on your path), delete your build dir, then rerun cmake with `-GNinja` instead of `-GMinGW Makefiles`.

nasihs, 2021-11-19:
> Hmm, this might be due to command line length limits on Windows, I've seen issues with that before. Try installing [ninja-build](https://ninja-build.org/) (just drop the exe somewhere on your path), delete your build dir, then rerun cmake with `-GNinja` instead of `-GMinGW Makefiles`.

Thanks for your attention.
I installed ninja and ran cmake -GNinja, ninja in the build directory, I got a ninja fatal error: Is the command too long?

[image]

[image]

Then I searched for the resolution and got this: https://stackoverflow.com/questions/43184251/cmake-command-line-too-long-windows
[image]

As I'm new to cmake, I don't know what the toolchain file specifies and I just added the settings in the root CMakeLists.txt like this:

[image]

I got a fatal error when I reran ninja. I checked the directory and as.exe did exist.

[image]

multiplemonomials, 2021-11-19:
Interesting, looks like that error is coming from gcc itself when it tries to execute the assembler.  Does the path in the red box actually exist on the disk?

Also, what happens if you use mbedignore (try the one in mbed-cmake-example-project) to cut the set of included mbed modules down to a lower number?  That should reduce the amount of flags being passed on the command line by a ton,

nasihs, 2021-11-22:
> Interesting, looks like that error is coming from gcc itself when it tries to execute the assembler. Does the path in the red box actually exist on the disk?
>
> Also, what happens if you use [...]

Thanks for your patience. The path in the red box exists. I try to use mbedignore, both mingw32 and ninja work well.

closed 2021-11-22 by nasihs