USCRPL/mbed-cmake #42: Add multiple projects in same workspace in vscode

jancotipsarevic, 2022-10-13:
Hi all, I´ve started working with mbed-cmake successfully after following the good documentation.
What I missed in the doc and I've tried by myself with no success is a commond feature found in others IDEs, which is to have many projects in the same workspace, with the capability of selecting the desired project then compile and flash the part.
I could imagine it is something that any of the contributors may do in the everyday working so it could be great to add some hints in the wiki to cover that point for the CMake beginners. Thank you very much.

multiplemonomials, 2022-10-13:
Do you mean, having multiple executables in the same project and then selecting between different ones to flash?

jancotipsarevic, 2022-10-13:
Hello Jamie, first of all let me congratulate you for the great mbed-cmake,
it is a great way to deal with mbed, custom targets etc..
What I meant is basically is described here  Multi-root Workspaces in
Visual Studio Code
<https://code.visualstudio.com/docs/editor/multi-root-workspaces> but
adapted to work with cmake-mbed.
Let say, to have a workspace (actually a folder) named with a specific
target board (for example) then any number of different cmake projects
(under its own directory). Then to have the ability to switch to any
project to compile and/or flash.
Now I have (the example projects created following your documentation):
-mbed_ws
----.vscode
----build
----mbed-cmake-example-project_1
----mbed-cmake-example-project_2

I can open the mbed_ws folder with vscode, I can switch between projects by
editing settings.json file into .vscode and type the desired project but
the only way to work with them after switch is to configure and compile the
whole project again.

Well I can open any simultaneous vscode projects and it is not a problem
for me but my guess is that experts guys have some way to achieve that.

Thank you very much

Miguel Angel Perez

multiplemonomials, 2022-10-15:
Unfortunately I haven't used this feature; I think you probably know as much about the situation as I do.  I'm much more of a CLion user, I'm still a bit of a noob with VS Code myself.

multiplemonomials, 2022-10-15:
Glad the project's working well for you though!  Make sure to also check out the newer version of mbed-cmake, [Mbed CE](https://github.com/mbed-ce)!  It's what I'm going to be updating going forward, and the VS Code integration works basically the same way as in this project.

jancotipsarevic, 2022-10-19:
It is ok, Thank you very much. Regarding Mbed CE I am already aware of this excellent initiative but in my case I am still considering adopting Mbed Os for our test tools in my company. So far we  have successfully developed a couple of tools based on nucleo boards with shields pcas. The problem arise using custom targets, I found some issues that make me feel it is not vey well supported (still in investigation), now I have a real case using STM32F103VDT that I managed to have it working in Mbed studio (in spite of having an error report saying "Compiler generates FPU instructions for a device whitout an FPU (check __FPU_PRESENT) and using ARMC6) but same custom target doesn't work in mbed-cmake in spite of compiling successfully (using gnu, may be the issue could be related to that), tha aplication is a simple blink led. Another rare aspect is with the generated PeriphreralPins.c (using STM32_gen_PeripheralPins.py to create it) the are some definitions that can not be found in Mbed Studio nor mbed-cmake, the AFIO_TIMx_ENABLE and others, I had to comment the related entries to be able to compile, but in my application I need to use pins PD_12 to PD_15 as pwms, see what I did 

`// TIM5 cannot be used because already used by the us_ticker
// (update us_ticker_data.h file if another timer is chosen)
MBED_WEAK const PinMap PinMap_PWM[] = {
    {PA_0,       PWM_2,  STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_NOPULL, 0, 1, 0)}, // TIM2_CH1
//  {PA_0,       PWM_5,  STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_NOPULL, 0, 1, 0)}, // TIM5_CH1
    {PA_1,       PWM_2,  STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_NOPULL, 0, 2, 0)}, // TIM2_CH2
//  {PA_1,       PWM_5,  STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_NOPULL, 0, 2, 0)}, // TIM5_CH2
    {PA_2,       PWM_2,  STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_NOPULL, 0, 3, 0)}, // TIM2_CH3
//  {PA_2,       PWM_5,  STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_NOPULL, 0, 3, 0)}, // TIM5_CH3
    {PA_3,       PWM_2,  STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_NOPULL, 0, 4, 0)}, // TIM2_CH4
//  {PA_3,       PWM_5,  STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_NOPULL, 0, 4, 0)}, // TIM5_CH4
    {PA_6,       PWM_3,  STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_NOPULL, 0, 1, 0)}, // TIM3_CH1
    {PA_7,       PWM_1,  STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_NOPULL, 6, 1, 1)}, // TIM1_CH1N
    {PA_7_ALT0,  PWM_3,  STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_NOPULL, 0, 2, 0)}, // TIM3_CH2
    {PA_7_ALT1,  PWM_8,  STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_NOPULL, 0, 1, 1)}, // TIM8_CH1N
    {PA_8,       PWM_1,  STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_NOPULL, 0, 1, 0)}, // TIM1_CH1
    {PA_9,       PWM_1,  STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_NOPULL, 0, 2, 0)}, // TIM1_CH2
    {PA_10,      PWM_1,  STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_NOPULL, 0, 3, 0)}, // TIM1_CH3
    {PA_11,      PWM_1,  STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_NOPULL, 0, 4, 0)}, // TIM1_CH4
    {PA_15,      PWM_2,  STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_NOPULL, 8, 1, 0)}, // TIM2_CH1
    {PB_0,       PWM_1,  STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_NOPULL, 6, 2, 1)}, // TIM1_CH2N
    {PB_0_ALT0,  PWM_3,  STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_NOPULL, 0, 3, 0)}, // TIM3_CH3
    {PB_0_ALT1,  PWM_8,  STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_NOPULL, 0, 2, 1)}, // TIM8_CH2N
    {PB_1,       PWM_1,  STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_NOPULL, 6, 3, 1)}, // TIM1_CH3N
    {PB_1_ALT0,  PWM_3,  STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_NOPULL, 0, 4, 0)}, // TIM3_CH4
    {PB_1_ALT1,  PWM_8,  STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_NOPULL, 0, 3, 1)}, // TIM8_CH3N
    {PB_3,       PWM_2,  STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_NOPULL, 8, 2, 0)}, // TIM2_CH2
    {PB_4,       PWM_3,  STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_NOPULL, 7, 1, 0)}, // TIM3_CH1
    {PB_5,       PWM_3,  STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_NOPULL, 7, 2, 0)}, // TIM3_CH2
    {PB_6,       PWM_4,  STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_NOPULL, 0, 1, 0)}, // TIM4_CH1
    {PB_7,       PWM_4,  STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_NOPULL, 0, 2, 0)}, // TIM4_CH2
    {PB_8,       PWM_4,  STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_NOPULL, 0, 3, 0)}, // TIM4_CH3
    {PB_9,       PWM_4,  STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_NOPULL, 0, 4, 0)}, // TIM4_CH4
    {PB_10,      PWM_2,  STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_NOPULL, 8, 3, 0)}, // TIM2_CH3
    {PB_11,      PWM_2,  STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_NOPULL, 8, 4, 0)}, // TIM2_CH4
    {PB_13,      PWM_1,  STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_NOPULL, 0, 1, 1)}, // TIM1_CH1N
    {PB_14,      PWM_1,  STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_NOPULL, 0, 2, 1)}, // TIM1_CH2N
    {PB_15,      PWM_1,  STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_NOPULL, 0, 3, 1)}, // TIM1_CH3N
    {PC_6,       PWM_3,  STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_NOPULL, 9, 1, 0)}, // TIM3_CH1
    {PC_6_ALT0,  PWM_8,  STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_NOPULL, 0, 1, 0)}, // TIM8_CH1
    {PC_7,       PWM_3,  STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_NOPULL, 9, 2, 0)}, // TIM3_CH2
    {PC_7_ALT0,  PWM_8,  STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_NOPULL, 0, 2, 0)}, // TIM8_CH2
    {PC_8,       PWM_3,  STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_NOPULL, 9, 3, 0)}, // TIM3_CH3
    {PC_8_ALT0,  PWM_8,  STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_NOPULL, 0, 3, 0)}, // TIM8_CH3
    {PC_9,       PWM_3,  STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_NOPULL, 9, 4, 0)}, // TIM3_CH4
    {PC_9_ALT0,  PWM_8,  STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_NOPULL, 0, 4, 0)}, // TIM8_CH4
[... cut ...]
    //{PE_12,      PWM_1,  STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_NOPULL, AFIO_TIM1_ENABLE, 3, 1)}, // TIM1_CH3N
    //{PE_13,      PWM_1,  STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_NOPULL, AFIO_TIM1_ENABLE, 3, 0)}, // TIM1_CH3
    //{PE_14,      PWM_1,  STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_NOPULL, AFIO_TIM1_ENABLE, 4, 0)}, // TIM1_CH4
    {NC, NC, 0}
};
`
If you have any hint to point to possible solution I´ll be pleased.
I will continue with the investigation and if needed I´ll open an issue .

Thank you very much
Regards

closed 2022-10-27 by jancotipsarevic