USCRPL/mbed-cmake #43: any plan to update to mbed-os 6.16?

jancotipsarevic, 2023-03-07:
Hi all, is there any plan to upgrade mbed-cmake to mbed-os 6.16? , I would like to use it with the recently added stm32 series STM32F103xD and STM32F103xG. I`ve tryed to copy mbed-os 6.16  into mbed-src (maintainig tools directory and CMakeLists.txt as noted in the doc) but it is plenty of errors.

Thank you.

jancotipsarevic, 2023-03-08:
Hello, this is what a get (target NUCLEO-F401RE, generator MinGW Makefiles, built by mingw32-make).

`[ 11%] Building C object mbed-cmake/mbed-src/CMakeFiles/mbed-os-static.dir/events/source/equeue_posix.c.obj
[ 11%] Building C object mbed-cmake/mbed-src/CMakeFiles/mbed-os-static.dir/events/tests/unit/prof.c.obj
C:\Mbed-Cmake\mbed_ws\mbed-cmake-example-project\mbed-cmake\mbed-src\events\tests\unit\prof.c: In function 'baseline_prof':
C:\Mbed-Cmake\mbed_ws\mbed-cmake-example-project\mbed-cmake\mbed-src\events\tests\unit\prof.c:46:5: error: impossible constraint in 'asm'
   46 |     __asm__ volatile ("rdtsc" : "=a" (a), "=d" (b));                        \
      |     ^~~~~~~
C:\Mbed-Cmake\mbed_ws\mbed-cmake-example-project\mbed-cmake\mbed-src\events\tests\unit\prof.c:56:24: note: in expansion of macro 'prof_cycle'
   56 |     prof_start_cycle = prof_cycle();                                        \
      |                        ^~~~~~~~~~
C:\Mbed-Cmake\mbed_ws\mbed-cmake-example-project\mbed-cmake\mbed-src\events\tests\unit\prof.c:126:9: note: in expansion of macro 'prof_start'
  126 |         prof_start();
      |         ^~~~~~~~~~
C:\Mbed-Cmake\mbed_ws\mbed-cmake-example-project\mbed-cmake\mbed-src\events\tests\unit\prof.c:46:5: error: impossible constraint in 'asm'
   46 |     __asm__ volatile ("rdtsc" : "=a" (a), "=d" (b));                        \
      |     ^~~~~~~
C:\Mbed-Cmake\mbed_ws\mbed-cmake-example-project\mbed-cmake\mbed-src\events\tests\unit\prof.c:60:23: note: in expansion of macro 'prof_cycle'
   60 |     prof_stop_cycle = prof_cycle();                                         \
      |                       ^~~~~~~~~~
C:\Mbed-Cmake\mbed_ws\mbed-cmake-example-project\mbed-cmake\mbed-src\events\tests\unit\prof.c:128:9: note: in expansion of macro 'prof_stop'
  128 |         prof_stop();
      |         ^~~~~~~~~
mingw32-make[2]: *** [mbed-cmake/mbed-src/CMakeFiles/mbed-os-static.dir/events/tests/unit/prof.c.obj] Error 1
mingw32-make[1]: *** [mbed-cmake/mbed-src/CMakeFiles/mbed-os-static.dir/all] Error 2
mingw32-make: *** [all] Error 2`

Thank you very much

multiplemonomials, 2023-03-09:
ahhhhh I remember this now.  There's a mistake in the original Mbed OS .mbedignore file.  Not sure how it was working before, but [this line](https://github.com/ARMmbed/mbed-os/blob/7e3ea686fbedec5b99dbebc9659df632ee121b12/events/tests/.mbedignore#L2) needs to be like [this](https://github.com/USCRPL/mbed-cmake/blob/ba2c51e368c078fa7121a46e54c122e8a88e1c4e/mbed-src/events/tests/.mbedignore#L2).

multiplemonomials, 2023-03-09:
Also, feel free to submit a PR to upgrade to Mbed 6.16!

closed 2023-03-13 by jancotipsarevic